sophgo/sophgo-doc #33: build warnings for TRM pdf

unicornx, 2023-10-08:
commit ID:  5d4b310

Run: `make latexpdf`

log indicates 5 warnings as below:

```
Running Sphinx v1.8.5
making output directory...
building [mo]: targets for 0 po files that are out of date
building [latex]: all documents
updating environment: 21 added, 0 changed, 0 removed
reading sources... [100%] todo                                                                                                                                 
/home/u/ws/sg2042/sophgo-doc/SG2042/TRM/source/SPI-flash.rst:136: WARNING: Title underline too short.

DMMR, Direct Memory Mapping Read (0x0C)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
/home/u/ws/sg2042/sophgo-doc/SG2042/TRM/source/SPI-flash.rst:136: WARNING: Title underline too short.

DMMR, Direct Memory Mapping Read (0x0C)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
/home/u/ws/sg2042/sophgo-doc/SG2042/TRM/source/SPI-flash.rst:332: WARNING: Unknown directive type "pic".

.. pic:: FmLen = 6, 1 byte per frame
/home/u/ws/sg2042/sophgo-doc/SG2042/TRM/source/SPI-flash.rst:334: WARNING: Unknown directive type "pic".

.. pic:: FmLen = 10, 2 bytes per frame
looking for now-outdated files... none found
pickling environment... done
checking consistency... done
processing SG2042TechnicalReferenceManual.tex...index abstract system pinmux clock Reset Power-domain LP PWM interrupt platform-control I2C SPI-flash GPIO todo 
resolving references...
/home/u/ws/sg2042/sophgo-doc/SG2042/TRM/source/clock.rst:188: WARNING: undefined label: platform_control (if the link has no caption the label must precede a section header)
writing... done
copying images... [100%] pic/DW_apb-and-Slave-Data-Widths.png                                                                                                  
copying TeX support files...
done
build succeeded, 5 warnings.
```

SophgoChaoWei, 2023-10-11:
dfac1f49314c6c1d0f784ee664504a56e5cb003f
fix this, thanks

closed 2023-10-11 by SophgoChaoWei

unicornx, 2023-10-11:
Thanks, verified and no warning now.