sophgo/sophgo-doc #69: Documentation for SCP firmware (fip.bin) for SG2042?

silvanshade, 2024-03-07:
The `fip.bin` is mentioned as the "SCP Firmware" for the SG2042:

1. https://github.com/sophgo/sophgo-doc/blob/e7f27be/SG2042/HowTo/bootflow.rst
2. https://github.com/sophgo/sophgo-doc/blob/215d2e6/SG2042/TRM/source/system.rst#bootrom

I see that the `fip.bin` is available here:

https://github.com/sophgo/bootloader-riscv/tree/64369b1/firmware

But I cannot find information about how `fip.bin` is generated or where it comes from.

Is the source for the firmware available?

Is there any information about how to re-implement a new custom `fip.bin`?

zamaudio, 2024-06-18:
@silvanshade I think the milkv folks are downstream of sophgo and their pioneer board is a custom implementation that uses the sg2042 chip.  We need to ask milkv to provide the board specific details to initialise their board. I have asked them to provide or upstream their source for bootloader-arm64 [here](https://github.com/milkv-pioneer/issues/issues/35) that compiles a fip.bin.

delgh1, 2024-11-24:
FWIW I also created an issue [here](https://github.com/milkv-pioneer/issues/issues/39) to get developers' attention.