sophgo/sophgo-doc #92: [suggestions] provide html build for the TRM

inochisa, 2024-03-22:
Although publishing pdf is more portable for users to use, html build have its own advantage:

1. html can be easily deployed on the web server.
2. html build can benefit from some feathers of sphinx, (e.g.: inline svg and wavedrom image)

As a result, I suggest to provide html tar for the TRM.

unicornx, 2024-03-22:
I default your are talking about sg200x and updated the titile of this issue. Let me know if anything wrong.

inochisa, 2024-03-22:
Hi, @unicornx. This suggestion is for all the document, even not just for SG2042 and SG200X, but the future SoCs.

unicornx, 2024-03-23:
adding assignment to @SophgoChaoWei, who is working on SG2042 documentation.

unicornx, 2024-04-13:
Seems no response from Sophgo on this new requirement for quite a long time, I will close it by now. We can reopen this when anyone raise this up again.

BTW, to my understand, TRM of SG200x and SG2042 both support building into html, you can try `make html` by yourself and get the html pages.
The only problem is, till now, only pdf version will be released out as official version.

closed 2024-04-13 by unicornx