sophgo/sophgo-doc #97: sg200x : trm: mmc/sd register missed P_VENDOR_SPECIFIC_AREA

unicornx, 2024-04-02:
[image]

这些 vendor 自己扩展的寄存器的起始地址从标准的 P_VENDOR_SPECIFIC_AREA 中读取，但是我们的 TRM中漏掉了 P_VENDOR_SPECIFIC_AREA

KennethLiu61, 2025-02-11:
之前没留意到这个问题。已经给到内部的owner确认是否有相关描述可以提供。应该今明两天能答复后确认是否可以关掉。

KennethLiu61, 2025-02-11:
P_VENDOR_SPECIFIC_AREA 默认是0x200，所以就没写出来。直接写了vendor相关寄存器。
相当于下图的效果：

[image]

unicornx, 2025-02-11:
明白了。你们对 trm 有什么改进建议吗？是保持原状，还是在什么地方补充说明一下你说的这个 “P_VENDOR_SPECIFIC_AREA 默认是0x200”。@KennethLiu61 

譬如在表格中增加一行，还是说在表格外面用文字说明一下？

KennethLiu61, 2025-02-11:
应该加一行描述就行了吧？就不改原来的excel表格了吧。

就是：
SG200X芯片设计的时候，给P_VENDOR_SPECIFIC_AREA配置了默认偏移地址 0x200 ，EMMC_CTRL = P_VENDOR_SPECIFIC_AREA + 0x0，EMMC_BOOT_CTL = P_VENDOR_SPECIFIC_AREA + 0x4 ,与表格对应以此类推。

PR #155 (2025-02-12): sg200x: trm: sdmmc: add note for registers
Fixed #97 

The starting address of the vendor extended register should be read from the standard P_VENDOR_SPECIFIC_AREA, but the description of P_VENDOR_SPECIFIC_AREA is missing in our TRM.

Solution: Add the description of the default value of P_VENDOR_SPECIFIC_AREA.
closed 2025-02-12 by unicornx