sophgo/sophgo-doc #122: DMA handshaking interface numbers are missing

nvitya, 2024-06-04:
Hi,

In the DMA channel CFG register there are two fields: DST_PER (bit pos 44) and SRC_PER (bit pos. 39). Also called as DMA hanshaking interface. They are probably 5-bit wide, and probably every peripheral require a specific number here. For example for the UART3_TX you need DST_PER=19 (i have tested this one). For SPI2_TX it is maybe 5 (i did not tes this yet).

This information is crucial to use the DMA properly. Could you add this to the reference manual?

Is there any source available somwhere with this information?

br.
nvitya

unicornx, 2024-06-04:
@nvitya what project are you talking about? SG2042 or SG200x?

nvitya, 2024-06-05:
@unicornx I'm testing on a SG2002 (using a Lichee RV Nano).

nvitya, 2024-06-06:
Hi,

I've wrote some test programs, it seems that using the DMA channel number for hardware handshaking interface works fine. Can you confirm that?

br.
nvitya