sophgo/sophgo-doc #131: sg200x:: the pinmux tables are incomplete

dzamlo, 2024-07-01:
For example, the pin XGPIOB[14] is missing.

unicornx, 2024-07-01:
> For example, the pin XGPIOB[14] is missing.

Can you please details the problem, in what product and in what section?  We have separated TRM for SG2002 and SG2000.

I double checked the document, seems the problem is in section "10.1.2 FMUX Registers Description" for SG2000, and it lacks some pin definition. I compare with the pin out xls sheet @ <https://github.com/sophgo/sophgo-hardware/blob/master/SG200X/03_SG2000/04_SG2000_PINOUT.xlsx>
Is this the problem?

dzamlo, 2024-07-02:
Yes this is the problem i'm talking about.

Also, I wasn't aware of the xls document. It may be worth it to mention it in the manual.

unicornx, 2024-07-02:
> Yes this is the problem i'm talking about.
> 
> Also, I wasn't aware of the xls document. It may be worth it to mention it in the manual.

Yes, actually the content of pinmux info in TRM is from that xls sheet, and seems something are missed for sg2000. 

Directly providing the link to xls doc in TRM may be a better idea, we are considering do like this.

unicornx, 2024-07-05:
Fixed with commit: a69a1c4.

We have released v1.01 to include this change.

https://github.com/sophgo/sophgo-doc/releases/tag/sg2002-trm-v1.01
https://github.com/sophgo/sophgo-doc/releases/tag/sg2000-trm-v1.01

closed 2024-07-05 by unicornx