sophgo/sophgo-doc #140: [SG200X] div_clk_187p5m, clk_187p5m registers missing, UART_SCLK configuration confusing

hellvesper, 2024-10-13:
There is a chapter about UART configuration in UART section page 555. UART_SCLK  configuration mention `div_clk_187p5m` register, which is missing in datasheet:
[image]
also there is mention about `clk_187p5m` register which description also missing:
[image]

Would be great to have an example for uart sclk configuration

unicornx, 2024-10-14:
@KennethLiu61 hi, can you please double check this?

KennethLiu61, 2024-10-17:
[image]
Sorry, there is an error in the document. This divider register should be div_clk_cam0_200.

closed 2024-10-28 by unicornx