sophgo/sophgo-doc #162: sg2002: Temperature sensor 的 channel 1 无法工作

heylenayy, 2025-02-25:
TRM 中描述 sg2002 中的 temperature sensor 有至少两路传感器通道，但按照手册的指引[1]同时使能 channel 0 与 channel 1 时，从 `tempsen_ch0_result` 和 `tempsen_ch1_result` 均不能读出有效数据，单独使能 channel 1 时也无法读出温度值。单独使能 channel 0 时温度值能够正常读取。

SDK 的代码中仅实现了对 channel 0 的读取[2]，请问温度传感器的硬件是否有相关的设计缺陷？

以及手册中提到芯片内建两个温度传感器[3]，但寄存器描述中却不时提到 channel 2 与 channel 3[4]，实际的通道数应该按照哪一个为准？

[1]: https://github.com/sophgo/sophgo-doc/blob/a9325c96dec34a3f4404e9e0065f9c4e03725e3b/SG200X/TRM/contents/cn/peripherals/temperature.rst?plain=1#L15-L28
[2]: https://github.com/sophgo/linux_5.10/blob/880032afe2ef765f2b4632c6e60cf35207064f4b/drivers/thermal/cv180x_thermal.c#L348-L366
[3]: https://github.com/sophgo/sophgo-doc/blob/a9325c96dec34a3f4404e9e0065f9c4e03725e3b/SG200X/TRM/contents/cn/peripherals/temperature.rst?plain=1#L13
[4]: https://github.com/sophgo/sophgo-doc/blob/a9325c96dec34a3f4404e9e0065f9c4e03725e3b/SG200X/TRM/contents/cn/peripherals/temperature_registers_description.table.rst?plain=1#L161-L162

unicornx, 2025-02-25:
@KennethLiu61 请帮忙 check 一下这个问题，谢谢

KennethLiu61, 2025-03-18:
收到，内部正在确认。