sophgo/sophgo-doc #166: SG2002: SD/MMC Host Controller Interface version issue

conkerkh, 2025-03-01:
Going through the source code of fsbl and u-boot, also looking at version reported by SD/MMC peripheral it seems that at least one of the controllers is at version 4.2. This is different than what is reported by the TRM which only talks about version 3.0 being supported. It has impact on the whole documentation as version 4 can support ADMA3 and UHS2 cards. This should be probably updated and reflected in the TRM

unicornx, 2025-03-02:
Thanks for your reporting.

Adding LiuGui, @KennethLiu61 would you please double check this?

KennethLiu61, 2025-03-17:
[image]
与内部对齐，SD是4.2而不是3.0，感谢这位同学发现的bug。
所以TRM中关于SD3.0需要改成SD4.2；其他不变。

[image]

unicornx, 2025-03-17:
OK, I will rasie a PR to fix this.